epics-modules/modbus #1: segmentation fault

MichaelRitzert, 2016-06-15:
I occasionally get a SIGSEGV in the modbus IOC. Now for the first time it happened when running in gdb. The backtrace is

> 2016/06/15 14:42:52.928 drvModbusAsyn::doModbusIO port block1316 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0 
>                                                                                        
> Program received signal SIGSEGV, Segmentation fault.                                   
> [Switching to Thread 0x7ffff42c3700 (LWP 315)]
> lockPort (pasynUser=0x0) at ../../asyn/asynDriver/asynManager.c:1677
> 1677        port     *pport = puserPvt->pport;
> (gdb) bt
> #0  lockPort (pasynUser=0x0) at ../../asyn/asynDriver/asynManager.c:1677
> #1  0x00007ffff7bd28ae in readPoller (pPlc=0x6ae7b0) at ../drvModbusAsyn.c:1319
> #2  0x00007ffff61fb07f in start_routine (arg=0x6b0d10)
>     at ../../../src/libCom/osi/os/posix/osdThread.c:385
> #3  0x00007ffff5379454 in start_thread (arg=0x7ffff42c3700) at pthread_create.c:334
> #4  0x00007ffff5677edd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

I will leave the gdb session open so that I can give more feedback as required.

MarkRivers, 2016-06-15:
The problem is happening on the call to pasynManager->lockPort.  The problem is that pPlc->pasynUserTrace is 0, i.e. a null pointer:

    while (1)
    {
        /* Sleep for the poll delay or waiting for epicsEvent with the port unlocked */
        pasynManager->unlockPort(pPlc->pasynUserTrace);
        if (pPlc->pollDelay > 0.0) {
            epicsEventWaitWithTimeout(pPlc->readPollerEventId, pPlc->pollDelay);
        } else {
            epicsEventWait(pPlc->readPollerEventId);
        }

        /* Lock the port.  It is important that the port be locked so other threads cannot access the pPlc
         * structure while the poller thread is running. */
        pasynManager->lockPort(pPlc->pasynUserTrace);
         
That asynUser structure is allocated in drvModbusAsynConfigure(), so it should not be 0.  Did this crash when the IOC first started up, or after it had been running for a while?  My first suspicion is that the pPlc structure has been corrupted by a memory overwrite.  What OS is this running on?  Can you use gdb to examine the entire pPlc structure (structure type = modbusStr).

MichaelRitzert, 2016-06-15:
It happens on Debian Linux after running OK for a while. Usually the IOC runs for days or weeks, today I just had two crashes within the first 15 minutes of running.

the pPlc structure:

> (gdb) up
> #1  0x00007ffff7bd28ae in readPoller (pPlc=0x6ae7b0) at ../drvModbusAsyn.c:1319
> 1319            pasynManager->lockPort(pPlc->pasynUserTrace);
> (gdb) p pPlc
> $1 = (PLC_ID) 0x6ae7b0
> (gdb) p *pPlc
> $2 = {portName = 0x0, octetPortName = 0x0, plcType = 0x0, isConnected = 0, 
>   ioStatus = 0, pasynUserOctet = 0x0, pasynUserCommon = 0x0, pasynUserTrace = 0x0, 
>   asynStdInterfaces = {common = {interfaceType = 0x0, pinterface = 0x0, 
>       drvPvt = 0x0}, drvUser = {interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     option = {interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, octet = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, octetProcessEosIn = 0, 
>     octetProcessEosOut = 0, octetInterruptProcess = 0, octetCanInterrupt = 0, 
>     octetInterruptPvt = 0x0, uInt32Digital = {interfaceType = 0x0, 
>       pinterface = 0x7ffff7ddac20 <drvUInt32D>, drvPvt = 0x6ae7b0}, 
>     uInt32DigitalCanInterrupt = 1, uInt32DigitalInterruptPvt = 0x6b0a00, int32 = {
>       interfaceType = 0x7ffff79aa4a6 "asynInt32", 
>       pinterface = 0x7ffff7ddabe0 <drvInt32>, drvPvt = 0x6ae7b0}, 
>     int32CanInterrupt = 1, int32InterruptPvt = 0x6b0a80, float64 = {
>       interfaceType = 0x7ffff79abefe "asynFloat64", 
>       pinterface = 0x7ffff7ddabc0 <drvFloat64>, drvPvt = 0x6ae7b0}, 
>     float64CanInterrupt = 1, float64InterruptPvt = 0x6b0b00, int8Array = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     int8ArrayCanInterrupt = 0, int8ArrayInterruptPvt = 0x0, int16Array = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     int16ArrayCanInterrupt = 0, int16ArrayInterruptPvt = 0x0, int32Array = {
>       interfaceType = 0x7ffff79aba3c "asynInt32Array", 
>       pinterface = 0x7ffff7ddaba0 <drvInt32Array>, drvPvt = 0x6ae7b0}, 
>     int32ArrayCanInterrupt = 1, int32ArrayInterruptPvt = 0x6b0b80, float32Array = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     float32ArrayCanInterrupt = 0, float32ArrayInterruptPvt = 0x0, float64Array = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     float64ArrayCanInterrupt = 0, float64ArrayInterruptPvt = 0x0, genericPointer = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, 
>     genericPointerCanInterrupt = 0, genericPointerInterruptPvt = 0x0, Enum = {
>       interfaceType = 0x0, pinterface = 0x0, drvPvt = 0x0}, enumCanInterrupt = 0, 
>     enumInterruptPvt = 0x0}, modbusSlave = 1, modbusFunction = 3, 
>   modbusStartAddress = 1316, modbusLength = 124, dataType = 0x6af650, 
>   data = 0x6af550, modbusRequest = "\001\003\005$\000|", '\000' <repeats 593 times>, 
>   modbusReply = "\003\370\000\000\000\000\000\000\000\000\000\000?\200\000\000\000\000\000\000B\310", '\000' <repeats 34 times>, "?\200\000\000A \000\000\000\000\000\000\000\000\000\000?\200\000\000\000\000\000\000B\310\000\000\000\000\000\000\000\000\000\000Ap\000\000@\320\000\000@\300\314\315@\f\000\000@\000\000\000\301\334", '\000' <repeats 14 times>, "A\020\000\000\000\000\000\000@\000\000\000@", '\000' <repeats 37 times>, "\231\232A!\000\000\000\000\231\232A!\231\232A!", '\000' <repeats 405 times>, 
>   pollDelay = 1, readPollerThreadId = 0x6b0d10, readPollerEventId = 0x6b06b0, 
>   forceCallback = 0, readOnceFunction = 0, readOnceDone = 0, readOK = 1715, 
>   writeOK = 0, IOErrors = 1, maxIOMsec = 1219, lastIOMsec = 18, timeHistogram = {
>     0 <repeats 200 times>}, histogramTimeAxis = {0 <repeats 200 times>}, 
>   enableHistogram = 0, histogramMsPerBin = 1, readbackOffset = 0}
>

MarkRivers, 2016-06-15:
This looks like a definite case of memory corruption, most of the pPlc structure is now 0, including pPlc->pasynUserTrace.  But it could not have been 0 previously, because the IOC and the modbus poller were running OK.  This is a difficult problem, because I suspect the issue is probably not in the modbus software itself, but some other software in your IOC that is overwriting this memory.  You should be able to use gdb put a watch on pPlc->portName, or pPlc->pasynUserTrace, and figure out what code is overwriting the pPlc structure.

MichaelRitzert, 2016-06-15:
I'll try to set up the watch.

The IOC I'm running is the plain modbusApp as it is built by the module build script.

MarkRivers, 2016-06-15:
What version of modbus and asyn are you running?

MichaelRitzert, 2016-06-15:
it's modbus git f2df7c782e279e26b21c34b7a3c5f8cb40e394b9
asyn asyn4-22

MichaelRitzert, 2016-06-15:
I should add: I also do see occasional crashes (segfault) on Scientific Linux 7 with asyn 4.29, modbus 2.8.
I haven't been able to investigate these in gdb yet, though.

Right now, the IOC is running fine while gdb is set up with the watches. It might crash while I type or in two weeks…

MarkRivers, 2016-06-15:
It looks like the memory corruption is only the first part of the pPlc structure.  It extends through at least the asynStdInterfaces/asynUInt32Digital/interfaceType element, but after that the data in the structure looks OK.

MichaelRitzert, 2016-06-16:
That didn't work… It crashed overnight, but before triggering the watchpoint first. Here's what I tried:

> (gdb) b readPoller
> Function "readPoller" not defined.
> Make breakpoint pending on future shared library load? (y or [n]) y
> Breakpoint 1 (readPoller) pending.
> (gdb) r UNICOS.cmd
> […]
> Breakpoint 1, readPoller (pPlc=0x6a9550) at ../drvModbusAsyn.c:1279
> 1279    {
> (gdb) watch pPlc->portName
> Watchpoint 2: pPlc->portName
> (gdb) watch pPlc->pasynUserTrace
> Watchpoint 3: pPlc->pasynUserTrace
> (gdb) disable
> (gdb) cont
> Continuing.

i.e. I set the watchpoint when readPoller was first entered. I think this is not correct, since during the crash, I have

> (gdb) p pPlc
> $1 = (PLC_ID) 0x6ae7b0

which is not at the same location at above.

How can I find the correct time to set the watchpoint?

MichaelRitzert, 2016-06-16:
Trying brute force. readPoller is entered x times, and I just set x watchpoints.
I also recompiled against asyn 4.29 for this attempt to rule out any long-gone bugs in my ancient asyn.

MarkRivers, 2016-06-16:
There is a separate pPlc structure for each drvModbusAsyn port driver you create, i.e. for each call to drvModbusAsynConfigure.  Do you know if it is always crashing on the same one?  You can tell from the portName or by modbus address and function code that particular port driver is using.  The safest thing is to put a watch on each one as you have done.

MarkRivers, 2016-06-16:
> readPoller is entered x times, 

x should be the number of times drvModbusAsynConfigure is called in your startup script.

MichaelRitzert, 2016-06-17:
got it.

> Watchpoint 4: pPlc->portName
> 
> Old value = 0x6ae3f0 "block1316"
> New value = 0x0
> Watchpoint 5: pPlc->pasynUserTrace
> 
> Old value = (asynUser *) 0x6b00e8
> New value = (asynUser *) 0x0
> getTraceMask (pasynUser=0x0) at ../../asyn/asynDriver/asynManager.c:2527
> 2527    {
> (gdb) bt
> #0  getTraceMask (pasynUser=0x0) at ../../asyn/asynDriver/asynManager.c:2527
> #1  0x00007ffff7bd241f in doModbusIO (pPlc=pPlc@entry=0x6aeb90, 
>     slave=<optimized out>, function=3, start=<optimized out>, data=0x6af930, 
>     len=<optimized out>) at ../drvModbusAsyn.c:1806
> #2  0x00007ffff7bd28d6 in readPoller (pPlc=0x6aeb90) at ../drvModbusAsyn.c:1322
> #3  0x00007ffff61f607f in start_routine (arg=0x6b12c0)
>     at ../../../src/libCom/osi/os/posix/osdThread.c:385
> #4  0x00007ffff5374454 in start_thread (arg=0x7ffff44e5700) at pthread_create.c:334
> #5  0x00007ffff5672edd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
> (gdb)

MarkRivers, 2016-06-17:
Unfortunately I'm not sure you did catch it.  It appears to me that gdb has caught the first read access of the new corrupt values.  But the code at the breakpoint cannot be the code that actually modified the values, it is the following:

    static int getTraceMask(asynUser *pasynUser)
    {
        userPvt  *puserPvt = asynUserToUserPvt(pasynUser);
        tracePvt *ptracePvt  = findTracePvt(puserPvt);

        return ptracePvt->traceMask;
    }

This code has simply been passed a null pointer from this code:

     asynPrintIO(pPlc->pasynUserTrace, ASYN_TRACEIO_DRIVER,
                        (char *)data, len*2,
                        "%s::doModbusIO port %s READ_COILS\n",
                        driver, pPlc->portName);
 
I don't know enough about gdb to know if it is possible to set a watch on a memory location for all write access, even writes that were not _supposed_ to write to that memory location.  I suspect it will slow things down a lot, and perhaps even cause the problem not to occur if it is a race condition or something.

MarkRivers, 2016-06-17:
The gdb documentation here
http://sourceware.org/gdb/download/onlinedocs/gdb/Set-Watchpoints.html#Set-Watchpoints
says

    Ordinarily a watchpoint respects the scope of variables in expr (see below). The -location 
    argument tells gdb to instead watch the memory referred to by expr. In this case, gdb will
    evaluate expr, take the address of the result, and watch the memory at that address. The type of
    the result is used to determine the size of the watched memory. If the expression's result does not
    have an address, then gdb will print an error.

So I think you need to add the --location flag when you set the watchpoint?

MichaelRitzert, 2016-06-17:
OK, I will try. Let's hope for a fast crash…

MichaelRitzert, 2016-07-04:
It wasn't exactly fast this time, but finally it did hit the watchpoint. And this time in a different location.
Looks like an off-by-one of some sort.

> Thread 15 "modbusApp" hit Hardware watchpoint 5: -location pPlc->portName
> 
> Old value = 0x6b0c20 "block1440"
> New value = 0x6b0000 "\001"
> 0x00007ffff7bd2405 in doModbusIO (pPlc=pPlc@entry=0x6b3a10, slave=<optimized out>, 
>     function=3, start=<optimized out>, data=0x6b1360, len=26)
>     at ../drvModbusAsyn.c:1804
> 1804                    data[i] = ntohs(pShortIn[i]);
> (gdb) bt
> #0  0x00007ffff7bd2405 in doModbusIO (pPlc=pPlc@entry=0x6b3a10, 
>     slave=<optimized out>, function=3, start=<optimized out>, data=0x6b1360, len=26)
>     at ../drvModbusAsyn.c:1804
> #1  0x00007ffff7bd28d6 in readPoller (pPlc=0x6b3a10) at ../drvModbusAsyn.c:1322
> #2  0x00007ffff61f607f in start_routine (arg=0x6b5d20)
>     at ../../../src/libCom/osi/os/posix/osdThread.c:385
> #3  0x00007ffff536f464 in start_thread (arg=0x7ffff41dc700) at pthread_create.c:334
> #4  0x00007ffff566de5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

I'll leave the debugger session open.

MichaelRitzert, 2016-07-04:
> (gdb) info locals
> readReq = <optimized out>
> readResp = 0x6b3ec8
> writeSingleReq = <optimized out>
> writeMultipleReq = <optimized out>
> readWriteMultipleReq = <optimized out>
> exceptionResp = <optimized out>
> requestSize = 6
> replySize = <optimized out>
> pCharIn = <optimized out>
> pCharOut = <optimized out>
> pShortIn = 0x6b3eca
> pShortOut = <optimized out>
> bitOutput = <optimized out>
> byteCount = <optimized out>
> status = asynSuccess
> i = 33
> startTime = {secPastEpoch = 836484360, nsec = 285667413}
> endTime = {secPastEpoch = 836484367, nsec = 488127613}
> nwrite = 6
> nread = 124
> eomReason = 1
> dT = <optimized out>
> msec = 7202
> bin = <optimized out>
> mask = 0 '\000'
> autoConnect = 1

MarkRivers, 2016-07-04:
If I understand correctly you have created the Modbus port block1480 with a length of 26.  But nread is coming back as 124, so it it looping from i=0 to 123.  It gets the access violation when i=33.  This makes sense, because data[] is only allocated 26 shorts.  So the question is why is nread=124, it should be at most 26.  

The pPlc structure you printed out has the values of the string that was send to the device, and the response that was received.  The string sent was:

"\001\003\005\310\000\032"

That string is this structure.

typedef struct modbusReadRequest_str
{
    unsigned char    slave;
    unsigned char    fcode;
    unsigned short   startReg;
    unsigned short   numRead;
} PACKED_STRUCTURE modbusReadRequest;

So slave=1, fcode=3, startReg=\005\310=1480, numRead=\000\032=26.  This all makes sense, it is exactly what I expect.

However, the response string in pPlc->modbusReply does not look correct:

modbusReply = "\003\370ff@\306\000\000\300\200\000\000\301\334\000\000\301\334\000\000\000\000\000\000\000\000\000\000?\200\000\000?

That string is this structure:

typedef struct modbusReadResponse_str
{
    unsigned char  fcode;
    unsigned char  byteCount;
    unsigned char  data[1];
} PACKED_STRUCTURE modbusReadResponse;

So fcode=3, which is correct.  However, byteCount=\370=248, which looks wrong.

This is the code where it is crashing:
        case MODBUS_READ_HOLDING_REGISTERS:
        case MODBUS_READ_INPUT_REGISTERS:
        case MODBUS_READ_INPUT_REGISTERS_F23:
            pPlc->readOK++;
            readResp = (modbusReadResponse *)pPlc->modbusReply;
            nread = readResp->byteCount/2;
            pShortIn = (epicsUInt16 *)&readResp->data;
            for (i=0; i<(int)nread; i++) { 
                data[i] = ntohs(pShortIn[i]);
            }
            asynPrintIO(pPlc->pasynUserTrace, ASYN_TRACEIO_DRIVER, 
                        (char *)data, nread*2, 
                        "%s::doModbusIO port %s READ_REGISTERS\n",
                        driver, pPlc->portName);
            break;

So nread=readResp->byteCount/2=124.  The code then loops from i=0 to i=123 and crashes when i=33 because it is getting out of array bounds on data[].

It appears to me that there are actually 2 problems:

- Your device appears to be sending an invalid reply, setting byteCount=254 when it should be 52.
- My driver is not protecting against the possibility that the returned byteCount is invalid, and is getting an access violation because it is exceeding the array bounds on data[].

I have made a change that checks the value of nread in the code above and if it is not equal to the requested length prints an error message and returns an error.  This branch is  wrong_read_size_fix on github.  I have tested that the branch compiles, but I don't have real Modbus system available to test actually running it.  Please test this new branch.  I predict that you will see occasional error messages when your device returns the wrong byteCount, but it should not crash anymore.

MarkRivers, 2016-07-04:
I was just able to do a quick test from Linux to a real Modbus device using function code 3, and the new code seemed to work OK.

MichaelRitzert, 2016-07-06:
Your fix works. I'm now running the branch version, and the crashes have been replaced with warnings. It always happens in timeout situations (the modbus connection is SSH-forwarded to another location, so these things can happen).

> 2016/07/05 17:49:30.206 drvModbusAsyn::doModbusIO port block1192 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:49:32.208 drvModbusAsyn::doModbusIO port block1316 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:49:34.210 drvModbusAsyn::doModbusIO port block1480 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:49:34.418 drvModbusAsyn::doModbusIO, port block1440 expected 25 words, actually received 62
> 2016/07/05 17:49:34.442 drvModbusAsyn::doModbusIO, port block1096 expected 62 words, actually received 124
> 2016/07/05 17:49:34.444 drvModbusAsyn::doModbusIO, port block1316 expected 124 words, actually received 26

Here the replies could have been mixed up. The number of received words is more or less correct, but the order is not.

> 2016/07/05 17:50:48.295 drvModbusAsyn::doModbusIO port block1316 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:50:50.298 drvModbusAsyn::doModbusIO port block1480 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:50:52.298 drvModbusAsyn::doModbusIO port block1440 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:50:54.300 drvModbusAsyn::doModbusIO port block1096 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:50:56.302 drvModbusAsyn::doModbusIO port block1192 error calling writeRead, error=127.0.0.1:5502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
> 2016/07/05 17:50:57.255 drvModbusAsyn::doModbusIO, port block1316 expected 124 words, actually received 26
> 2016/07/05 17:50:57.269 drvModbusAsyn::doModbusIO, port block1480 expected 26 words, actually received 25
> 2016/07/05 17:50:57.283 drvModbusAsyn::doModbusIO, port block1440 expected 25 words, actually received 26
> 2016/07/05 17:50:57.286 drvModbusAsyn::doModbusIO, port block1096 expected 62 words, actually received 25

This is most certainly just nonsense.

MarkRivers, 2016-07-06:
OK, that's good.  You didn't say previously that you were getting timeouts.  That makes it more understandable.  The driver does an atomic flush/write/read operation, which should guarantee that each Modbus port driver gets the reply to its own command, and not some other port driver's command.  However, if there is a timeout then a reply to a previous command could be received after the flush operation, so the incorrect reply to a previous command may be received.  The driver now protects against this.

MichaelRitzert, 2016-07-08:
There happens to be a timeout in the very first message. However without the patch, I hadn't been able to positively confirm a correlation between the timeout messages (with timestamp) and the crash (no timestamp).
I've been running the patched version since Tuesday, and I have seen the new log message but no more crashes so far.

closed 2016-07-08 by MichaelRitzert