epics-modules/modbus #6: Modbus driver continuously prints error messages when the device is disconnected

MarkRivers, 2016-11-28:
The Modbus driver continuously prints error messages when the device is disconnected.  It should just put the error string in pasynUser->errorMessage and pass it back to device support.

slacAWallace, 2016-11-28:
2016/11/28 11:47:19.160 drvModbusAsyn::doModbusIO port aiFLOAT_PORT Modbus exception=4
2016/11/28 11:47:19.163 drvModbusAsyn::doModbusIO port aiLONG_PORT Modbus exception=4
2016/11/28 11:47:19.164 drvModbusAsyn::doModbusIO port aoLONG_PORT Modbus exception=4
2016/11/28 11:47:19.165 drvModbusAsyn::doModbusIO port BI_PORT Modbus exception=4
2016/11/28 11:47:19.262 drvModbusAsyn::doModbusIO port AI_PORT Modbus exception=4
2016/11/28 11:47:19.263 drvModbusAsyn::doModbusIO port aiFLOAT_PORT Modbus exception=4
2016/11/28 11:47:19.265 drvModbusAsyn::doModbusIO port aiLONG_PORT Modbus exception=4
2016/11/28 11:47:19.267 drvModbusAsyn::doModbusIO port aoLONG_PORT Modbus exception=4
2016/11/28 11:47:19.269 drvModbusAsyn::doModbusIO port BI_PORT Modbus exception=4
2016/11/28 11:47:19.365 drvModbusAsyn::doModbusIO port AI_PORT Modbus exception=4
2016/11/28 11:47:19.367 drvModbusAsyn::doModbusIO port aiFLOAT_PORT Modbus exception=4
2016/11/28 11:47:19.369 drvModbusAsyn::doModbusIO port aiLONG_PORT Modbus exception=4
2016/11/28 11:47:19.370 drvModbusAsyn::doModbusIO port aoLONG_PORT Modbus exception=4
2016/11/28 11:47:19.371 drvModbusAsyn::doModbusIO port BI_PORT Modbus exception=4
2016/11/28 11:47:19.468 drvModbusAsyn::doModbusIO port AI_PORT Modbus exception=4
2016/11/28 11:47:19.469 drvModbusAsyn::doModbusIO port aiFLOAT_PORT Modbus exception=4
2016/11/28 11:47:19.471 drvModbusAsyn::doModbusIO port aiLONG_PORT Modbus exception=4
2016/11/28 11:47:19.473 drvModbusAsyn::doModbusIO port aoLONG_PORT Modbus exception=4
2016/11/28 11:47:19.475 drvModbusAsyn::doModbusIO port BI_PORT Modbus exception=4

I think the host is reachable, but maybe the modbus server isn't running.

MarkRivers, 2016-11-29:
This is not the error I was expecting to see.  This error should indicate that the Modbus server responded but is reporting an error, which is called an exception in Modbus terminology.  

This is a link to a table of Modbus exception codes: [https://www.kepware.com/en-us/support/resource-library/legacy-technical-notes/modbus-exception-codes/](exception table).  Note that exception 4  is a "Failure In Associated Device".  

Can you check whether the Modbus server is actually running, but unable to talk to the hardware or something?

slacAWallace, 2016-11-29:
You’re right, this was not the error that is filling up our logs.

This is the one. I simulated the PLC being disconnected by triggering a power-cycle.

2016/11/28 17:15:43.871 drvModbusAsyn::doModbusIO port aoLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:45.871 drvModbusAsyn::doModbusIO port BI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:47.870 drvModbusAsyn::doModbusIO port aiFLOAT_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:49.871 drvModbusAsyn::doModbusIO port aiLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:51.872 drvModbusAsyn::doModbusIO port AI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:53.873 drvModbusAsyn::doModbusIO port aoLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:55.873 drvModbusAsyn::doModbusIO port BI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:57.873 drvModbusAsyn::doModbusIO port aiFLOAT_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:15:59.872 drvModbusAsyn::doModbusIO port aiLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:01.872 drvModbusAsyn::doModbusIO port AI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:03.872 drvModbusAsyn::doModbusIO port aoLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:05.872 drvModbusAsyn::doModbusIO port BI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:07.872 drvModbusAsyn::doModbusIO port aiFLOAT_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:09.873 drvModbusAsyn::doModbusIO port aiLONG_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0
2016/11/28 17:16:11.873 drvModbusAsyn::doModbusIO port AI_PORT error calling writeRead, error=plc-mfx-sds:502 timeout: Resource temporarily unavailable, nwrite=6/6, nread=0

shadowguy, 2017-08-02:
Hi guys,

I've been taking a look at this, and based on Mark's initial writeup and some sample EPICS code I found around the net, I propose the following code change in modbusApp/src/drvModbusAsyn.c.

From:
    if (status != asynSuccess) {
        asynPrint(pPlc->pasynUserTrace, ASYN_TRACE_ERROR,
                 "%s::doModbusIO port %s error calling writeRead,"
                 " error=%s, nwrite=%d/%d, nread=%d\n", 
                 driver, pPlc->portName, 
                 pPlc->pasynUserOctet->errorMessage, (int)nwrite, requestSize, (int)nread);
        pPlc->IOErrors++;
        goto done;
    }

To:
    if (status != asynSuccess) {
        epicsSnprintf(pPlc->pasynUserCommon->errorMessage,
                 pPlc->pasynUserCommon->errorMessageSize,
                 "%s::doModbusIO port %s error calling writeRead,"
                 " error=%s, nwrite=%d/%d, nread=%d\n",
                 driver, pPlc->portName,
                 pPlc->pasynUserOctet->errorMessage, (int)nwrite, requestSize, (int)nread);
        /* Note if this has happened, there's a good chance the target is offline */
        status = asynError;
        pPlc->IOErrors++;
        goto done;
    }

Tested and it builds.  Not sure about the pasynUser - should it go to pasynUserTrace instead?  Am I missing anything?

MarkRivers, 2017-09-15:
I think it is important that there be at least one error message printed when the IOC loses connection to the device.  Simply passing an error status to asyn device support for I/O Intr scanned records does not print any message at all.  

In R2-10 I have change the code so it prints a single message for each modbus asyn driver when the status from doModbusIO changes, either from success to failure, or failure to success.  This way there is a single message when the device become unavailable, and another when it reconnects.

closed 2017-09-15 by MarkRivers