epics-modules/modbus #7: MODBUS/TCP does not make use of transaction ID

norumwe12, 2017-01-19:
The MODBUS/TCP transaction ID is set to 1 on writes and ignored on reads.  It would be more robust if a unique ID were used for each transaction and replies with invalid IDs ignored -- especially for MODBUS/UDP links.  My proposed change is attached.

[modbusTIDpatch.txt](https://github.com/epics-modules/modbus/files/717562/modbusTIDpatch.txt)

Engbretson, 2017-01-19:
You want a unique transaction ID but only allocated space for a short?  Isn’t this going to overflow or wrap around/overflow like 5 minutes or less from now if each request over the network requires a new id?

norumwe12, 2017-01-19:
Yes, the MODBUS spec provides for only a 16 bit value which will wrap around after a while.  “5 minutes or so” is unlikely, though — that would imply over 200 transactions per second.

But even a 16 bit ID better than nothing.  It’s unlikely that a duplicate packet will show up after more than a few seconds so even a five minute wraparound interval will be fine.

> On Jan 19, 2017, at 1:26 PM, Engbretson <notifications@github.com> wrote:
> 
> You want a unique transaction ID but only allocated space for a short? Isn’t this going to overflow or wrap around/overflow like 5 minutes or less from now if each request over the network requires a new id?

-- 
Eric Norum
wenorum@lbl.gov

MarkRivers, 2017-09-15:
Closed with #8 .  In R2-10-1.

closed 2017-09-15 by MarkRivers