epics-modules/modbus #31: Support for modbus function 17 (Feature request)

nusaqib, 2023-12-19:
We are in the process of migrating LabVIEW to EPICS for beamline equipment protection system that consists of Modicon PLCs. All PVs are translated except status registers that use modbus function 17. It would be great if the module has support for this.

Reference: 
- [Modicon Modbus Protocol Reference Guide - Page 48, 49, 54](https://modbus.org/docs/PI_MBUS_300.pdf)
- [Modicon 984 series manual](https://pco2.com/v/vspfiles/downloadables/Modicon/984_systemmanual.pdf)

Thanks,
Najm

MarkRivers, 2023-12-19:
The documentation for function 17 in the Modbus specification is on page 32 of this document:
https://epics-modules.github.io/modbus/_downloads/c7a9377b51990ac029036109bef240f5/Modbus_Application_Protocol_V1_1b.pdf

It says that function 17 is only supported for serial line connections, not for TCP.

The response to the function 17 query is the following:
- The function code (17) (1 byte)
- The byte count (1 byte)
- The slave ID (device specific, but presumably just 1 byte?)
- The Run Indicator status (1 byte, 0x00=Off, 0xFF=On)
- Additional data (device dependent)

The additional data for various Modicon models is:
- Model 184 and 384, 72 bytes
- Model 584 and 984, 7 bytes
- Model 84, 2 bytes
- Model 484 and 884, 3 bytes

This could be implemented by reading the response into a byte array which is copied to a waveform record with FTVL=SHORT and Modbus datatype "UINT16".

I do not have any PLCs with serial line interfaces, and I don't have any Modicon PLCs at all, so I cannot test this.

You are welcome to modify the driver to handle function 17, and I am happy to review a pull request.

nusaqib, 2024-01-17:
Hi @MarkRivers, does 'Serial Line only' implies that it is not possible to use an ethernet to serial adapter for this? I tried to test via python and a serial server, but as soon as I select modbus TCP, I get 'function not supported'.

MarkRivers, 2024-01-17:
You should be able to use a terminal server.  You should not use Modbus TCP, you should select Modbus RTU or Modbus ASCII depending on what your Modicon PLC supports.  What hardware have you been using to talk to the Modicon from LabView?  Is the Modicon an RS-485 or RS-232 connection?

nusaqib, 2024-01-17:
For LabView, currently there is a serial cable (RS-232) connected directly to a server running LabView application. I want to get rid of this in case of EPICS and want to use a terminal server.

Regarding [python pymodbus](https://pymodbus.readthedocs.io/en/latest/source/readme.html), what I understand is there are only two client options: Serial for a serial port and TCP for ethernet, both of these then have a framer option to use RTU or ASCII. I tried using TCP client and RTU framer, but gives 'not supported' error.

MarkRivers, 2024-01-17:
In EPICS you can define an asyn TCP port to the terminal server, but you can use Modbus RTU or Modbus ASCII as the protocol.  Were you using Modbus RTU or ASCII with LabView?

MarkRivers, 2024-01-17:
The packet you need to send for F17 is very simple.  You can just build that packet in a little Python program, send it to the PLC via the terminal server, and then read the response into a buffer that you can print with Python.  That should be good enough for the proof of concept.

nusaqib, 2024-01-17:
Thanks Mark, I'll try that.

nusaqib, 2024-01-18:
I am able to read response for function 17 via terminal server using python and also EPICS stream device. Hopefully, will try to implement in the modbus driver and do a pull request.

Thanks for your guidance.

MarkRivers, 2024-01-18:
I will modify the driver so it should support function 17.  I will need to to test it.

nusaqib, 2024-01-18:
Thank you Mark! Sure thing, let me know and I will test it.

nusaqib, 2024-01-22:
Hi @MarkRivers, I did a pull request for function 17, please review and make changes accordingly. It worked fine.

closed 2024-01-22 by nusaqib