epics-modules/modbus #35: Encounter a problem using Modbus

Baokangwen, 2025-06-10:
Dear @MarkRivers 

##Load record instances
dbLoadRecords("/home/guest/EPICS/synApps/support/modbus-R2-11/db/18ubraker.db")
Can't set "18U:TABLE:DETH-DI0:DI.DTYP" to "asynUInt32Digital"
Error at or before ")" in file "/home/guest/EPICS/synAppa/support/modbus-R2-11/db/18ubraker.db"

ErrordbLoadRecords: faild to load /home/guest/EPICS/synAppa/support/modbus-R2-11/db/18ubraker.db

MarkRivers, 2025-06-11:
> Can't set "18U:TABLE:DETH-DI0:DI.DTYP" to "asynUInt32Digital"

Your db file looks OK.  I think the problem must be that your IOC application does not correctly include the asyn dbd file, so it does not know about the asynUInt32Digital device support.

Baokangwen, 2025-06-12:
Thank you mark.

MarkRivers, 2025-06-13:
Is it working now?

Baokangwen, 2025-06-20:
I am still trying, there may be some other issues, thank you mark.

MarkRivers, 2026-09-07:
I have not heard back about this so I am closing the issue.

closed 2026-09-07 by MarkRivers